wake-0/fhvRobot #3: [Communication] [Server] [Robot] Specification of communication protocol

trylimits, 2015-10-29:
We should discuss and specify a protocol for the communication between Server and Robot.

Up to layer 4 of OSI we use UDP/IP protocol.

Upper layers should be specified.

trylimits, 2015-11-19:
We have defined the protocol as follows:

Session layer enriches as follows:

`Flags(1B) SessId(1B) Message`

Presentation layer enriches as follows:

`Flags(1B) Message`

Application layer enriches as follows:

`Flags(1B) CommandID(1B) Len(1B) Payload`

whereby Bits 0 and 1 are used for version for each `Flag` byte.

In the application layer Bits 6 and 7 are used to indicate which device is used (`00` Robo, `01` App).
In the session layer Bit 7 is used to indicate whether the message represents a session request (`1` if a session id is requested)

trylimits, 2015-11-19:
Someone should document the protocol definition in a proper form and then close this issue.

Blackjack92, 2015-11-23:
It is necessary to add a name or id for the robos otherwise it is not possible to know which robo should handle the command (only when more than one robo is used).

trylimits, 2015-11-23:
Is this really necessary? In my opinion it should be this way:

1. Robot establishes session (already done)
2. Robot registers at the server with its name. See #16 
3. Server has to distinguish between the robots with a mapping between session and name

The sequence in the server's code should be as follows:

1. Open the socket
2. Session layer is responsible for handling session establishment (no interaction with application layer at all, as already implemented)
2.1. Session layer is responsible for a mapping between session id and the actual receiver's address.
3. When robot registers with #16 command your application layer gets a callback, e.g. `void messageReceived(int sessionId, byte[] message)` which allows you to implement a mapping between name and session id
4. Send commands with an API like for example `send(int sessionId, byte[] message)`

One might think that the application layer (or any other upper layer) should not know about a session id, however, this is necessary in some way as it would not be possible for the server's session layer to send a message to the right receiver.
Am I missing something?

If we would add a field for the name, would a robot called "Alice" ever get a command for a robot called "Bob"? Would the robot have to check if the command corresponds to itself, i.e. if "Alice" gets a command for "Bob" it simply would ignore it?
On the Robot's side I thought the session layer would just check the session id (discarding/ignoring messages with an invalid session id) and then forward the message to the upper layers.

Blackjack92, 2015-11-30:
The session protocol was changed. For requesting a session send:
Flags(1B) SessId(1B) Message
0000 0001 | 0000 0000 | ....
The position of the request bit was changed, easier to use in java. The max number of clients is defined as 127 (1 to 127).

Blackjack92, 2015-12-05:
Adding a packet size byte to the network layer would be very good so it is possible to easily cut the byte which are unused.

trylimits, 2015-12-06:
@Blackjack92 We already have a payload size (length) byte in the application layer. Do you meant something different?

Blackjack92, 2015-12-07:
I mean that we add a packet size to the network layer so that we know what is the payload of the network layer packet. otherwise it is not easy to determine how long the received packet is.

Blackjack92, 2015-12-07:
All other protocol related stuff should be added to #26 and/or the wiki itself.

closed 2015-12-07 by Blackjack92