wake-0/fhvRobot #7: [Server] Add media streaming

Blackjack92, 2015-11-02:
Add a prototype for streaming a video over the network.

trylimits, 2015-11-03:
I did some research about streaming from BBB to another machine by using cvlc.

First important note:

VLC does not run out of the box on bbb (bc of root user). Solution to that is to change the executable file of vlc: http://www.hackersgarage.com/vlc-is-not-supposed-to-be-run-as-root-sorry-solution.html

Then I tried those examples for streaming:
https://sandilands.info/sgordon/live-webca-streaming-using-vlc-command-line

However, all experiments have to be repeated as I did not document which commands worked or worked not.

If we have found a command which is capable of streaming media to the server we should set up shell-scripts which runs these commands which we then can execute from code. I think this concept is pretty neat for our issue, as it allows to replace streaming service without changing any code (full decoupling) and should be straightforward to implement.

To summarize the steps we have to do to resolve this issue:

1. Experiment/test and document which command streams media to the server
2. Set up shell script for that command
3. Set up shell script to stop streaming
4. Run these shell scripts from Robo-Code

If we have done these steps successfully we simply can redistribute the stream from the server to the clients. This should not be hard to implement as it is simply "bridging" the data from one socket to another.

mpe5651, 2015-11-12:
I'll take care of this topic

mpe5651, 2015-12-04:
@trylimits as you are the linux guy in our team, I would like to ask you for help and show you my problem with the video4linux command, preferably tomorrow (05.12.2015). is that possible?

Blackjack92, 2015-12-04:
I think this would be very good because I am implementing the server and app stuff at the moment.

trylimits, 2015-12-05:
To get internet working on BBB use this command: `/sbin/route add default gw 192.168.7.1`
`

mpe5651, 2015-12-06:
after trying several options the following observations were made:

1.) tutorial from DM ist not working with our webcam (missing H264 encoder support)
2.) all `cvlc` commands were not satisfying
    2.1) the only output format which was working was ogg
3.) this guy https://github.com/Mras2an/Webcam-IP-BeagleBone made a piece of code which allows us to establish a working udp video stream which is receivable by the VLC player

trylimits, 2015-12-07:
To complete @mpe5651 's post above:
We fixed the latency problem in the end and have nearly no delay (<100ms) atm. (resolution 320x240, with color, no sound)! This is perfect and a good basis for future work!

However, we did not get the stream to the server (using VLC on the server as client and using the public IP address at port 995 of the stream as target on the BBB). Two questions:
1. @Blackjack92 Did you explicitly open ports 997/998 for communication. Could it be that the firewall of the server blocked incoming data of port 995?
2. Could it be, that FHV blocks UDP network traffic with a specific packet size? Communication packets have 1024B atm. which is working. Stream packets have approx. 50kB atm and are not working.

Nevertheless, we have to fix above problem and then we are fine with media streaming for the moment. Probably it will just work out of the box if Ticket #30 (WIFI dongle) is used instead.

Moreover, I did additional research yesterday evening and we have to keep some things in mind:

1. We can definitely not use H264 with our webcam (as @mpe5651 already described in 1. above)
2. We won't get UDP multicast run in our environment so we have to distribute the stream to all connected clients independently (shouldn't be a problem in terms of software, but may be a problem in terms of network traffic/bandwith)
3. Because of 2. highest priority is to minimize network traffic (packet size) and maybe we should stick to b/w stream only (at least we will have to implement an option to change that remotely)

trylimits, 2015-12-07:
I just did a speed test on the server and I have good news:

[image]

This bandwidth should satisfy our requirements for distributing the stream to at least 10 clients.

trylimits, 2015-12-25:
I did some further research and testing on that topic. These are the findings:

1. I had problems to get the stream working using the @mpe5651 his program of 3.) on my own machine. It seems that the firewall blocks the traffic and VLC is not able to receive the stream. The solution is to add an exception to the firewall for vlc.exe. **I already did that on the server**

2. Having a frame size of 320x240@30fps consumes 4MB traffic per second which is way too much. Reducing to 160x120@10fps reduces the traffic to 300kB/s which is also too much for my home internet connection, but should be good enough for FHV/server network. Longterm, we should change the code of the streaming program to transmit one keyframe a second and then diff frames only. This should fairly decrease the bandwidth and we could go up to 640x480@30fps or more. The implementation should be straightforward, but should be implemented with low priority. Steps would be to select different frame types when fetching frames from v4l driver. See http://linuxtv.org/downloads/v4l-dvb-apis/buffer.html#v4l2-buf-type

3. On the server side we can use VLCJ (https://github.com/caprica/vlcj) and transpose the udp stream to a http stream which then can be shown with JavaFX and Android natively. VLCJ is very easy to use and is able to do all that out of the box. However, I was not able to get VLC 64Bit in combination with JDK 64Bit to work. I guess we have to stick to VLC 32Bit and also change to JDK 32Bit.

trylimits, 2016-01-03:
I found a way to decrease network traffic dramatically. Inspecting the sent packets showed that the major part of the frame's data are trailing zeros after MJPEG's end segment `0xFF` `0xD8`. Getting rid of these zeros results in packet sizes of about a tenth or a fifth of the current packet sizes.

trylimits, 2016-01-04:
Above commit implements media streaming on the server. Additional functionality should be handled in another ticket.

closed 2016-01-04 by trylimits