wake-0/fhvRobot #15: [App] Implement communication layers

trylimits, 2015-11-19:
This ticket is about the implementation of the communication layers for the Android App.

@Blackjack92 Could you please evaluate if it is possible to completely reuse the communication code of the Server implementation for the App?

Another good option would be if @mpe5651 takes care about the whole app stuff, after a short introduction into the Android SDK. In that case all team members would be responsible for one component.

Blackjack92, 2015-11-19:
I already started with the app and pulled out the communication stuff in an own project so that it is possible to reuse. I think it would be best when @mpe5651 focuses on the video streaming and only helps with the app.
Some minor changes will be needed so that it is possible for deciding if it is a server or client who communicates. depending on this the communication is different.

trylimits, 2015-11-19:
Good work.

Blackjack92, 2015-12-03:
This should be working, when something is not correct, open a bug ticket.

closed 2015-12-03 by Blackjack92