wake-0/fhvRobot #16: [Command] [Server] [Robo] Register Robot

trylimits, 2015-11-19:
This ticket is about the implementation of the following command:

@trylimits
From: Robot
To: Server
Command ID: 1
Len: [1, 255] (0 should not be accepted by the server)
Payload: The actual name of the Robot acc. Len (not null-terminated)

The Server's application layer should response:

@Blackjack92 
From: Server
To: Robot
Command ID: 1
Len: 1
Payload: 1 if accepted, 0 otherwise (duplicate name, too much robots, ...)

Raw package (example):

`0000 0001    0000 0011    ... three characters ...`

The preceding `Flag` byte should be acc. specification, i.e. `0x00` in this case. 

The Robot's policy:
1.) No commands are accepted until registered at the server
2.) No commands are accepted until registering was accepted by the server
3.) After registering was accepted the Robot should be in remote operation mode

Blackjack92, 2015-11-23:
Please try this registration (changing name of an existing robo).

trylimits, 2015-12-06:
This seems to work properly. Closing for now.

closed 2015-12-06 by trylimits