wake-0/fhvRobot #17: [Command] [Server] [Robo] Drive Motors

trylimits, 2015-11-19:
This ticket is about the implementation of the following command:

@Blackjack92 
From: Server
To: Robot
Command ID: `10` for right, `11` for left, `12` for both motors
Len: [1] (any other value should not be accepted by the Robot)
Payload: The actual value as signed byte (`-100 to 100`)

@trylimits
The Robot's policy should be:

1. Set the motor speed to the given value
2. If no motor command for >5s is received the Robot should stop (emergency mode)

Blackjack92, 2015-11-23:
Please verify this functionality. There are three buttons on the server application.

trylimits, 2015-12-06:
Works properly from the server and from the app. Not committed yet, as some clean-ups have to be made. Ticket #34 still relevant!

trylimits, 2015-12-11:
Implemented command. Solved.

trylimits, 2015-12-13:
On App- and Server-side the commands for controlling left/right motor are reversed.

We have to fix that and recompile the library.

trylimits, 2015-12-13:
Fixed.

closed 2015-12-13 by trylimits