wake-0/fhvRobot #21: [Server] Bug in sending responses

trylimits, 2015-12-03:
At the moment all responses sent from the Server to the Robot are of length 1024, even though the actual data is only 2 bytes long. The sent message should be exactly as long as it actually is, otherwise it will not be possible to unpack the data on the Robot's side.

trylimits, 2015-12-03:
I already tracked down the problem. An efficient way to solve this issue to let the sender define the length of the message, which makes most sense to me. These would be the changes:

```
diff --git a/Source_Server/RoboServer.Network/src/communication/managers/IAnswerHandler.java b/Source_Server/RoboServer.Network/src/communication/managers/IAnswerHandler.java
index 21112e3..08bc984 100644
--- a/Source_Server/RoboServer.Network/src/communication/managers/IAnswerHandler.java
+++ b/Source_Server/RoboServer.Network/src/communication/managers/IAnswerHandler.java
@@ -15,7 +15,7 @@ import communication.IConfiguration;
 
 public interface IAnswerHandler {
 
-	public void answer(IConfiguration configuration, byte[] data);
+	public void answer(IConfiguration configuration, byte[] data, int len);
 
 	public void answer(IConfiguration configuration, DatagramPacket datagram);
 }
diff --git a/Source_Server/RoboServer.Network/src/communication/managers/SessionManager.java b/Source_Server/RoboServer.Network/src/communication/managers/SessionManager.java
index 5338bbd..14f9e6d 100644
--- a/Source_Server/RoboServer.Network/src/communication/managers/SessionManager.java
+++ b/Source_Server/RoboServer.Network/src/communication/managers/SessionManager.java
@@ -58,7 +58,7 @@ public class SessionManager extends LayerManager<SessionPDU> {
 			answer[1] = newByteSession;
 
 			// Send answer pdu
-			sender.answer(currentConfiguration, answer);
+			sender.answer(currentConfiguration, answer, 2 /* Length of answer */);
 
 			// Everything is handled so no need to go to the upper layers
 			handled = true;
diff --git a/Source_Server/RoboServer.TestClient/src/UDPClient.java b/Source_Server/RoboServer.TestClient/src/UDPClient.java
index d0f4796..5bfea82 100644
--- a/Source_Server/RoboServer.TestClient/src/UDPClient.java
+++ b/Source_Server/RoboServer.TestClient/src/UDPClient.java
@@ -100,7 +100,7 @@ public class UDPClient
 	}
 
 	@Override
-	public void answer(IConfiguration configuration, byte[] data) {
+	public void answer(IConfiguration configuration, byte[] data, int len) {
 
 	}
 
diff --git a/Source_Server/RoboServer/src/application/fhv/network/Communication.java b/Source_Server/RoboServer/src/application/fhv/network/Communication.java
index 4395c55..03803b3 100644
--- a/Source_Server/RoboServer/src/application/fhv/network/Communication.java
+++ b/Source_Server/RoboServer/src/application/fhv/network/Communication.java
@@ -93,7 +93,7 @@ public class Communication implements Runnable, IDataReceivedHandler<Application
 	}
 
 	@Override
-	public void answer(IConfiguration configuration, byte[] data) {
+	public void answer(IConfiguration configuration, byte[] data, int len) {
 		InetAddress address;
 		try {
 			address = InetAddress.getByName(configuration.getIpAddress());
@@ -103,7 +103,7 @@ public class Communication implements Runnable, IDataReceivedHandler<Application
 		}
 
 		int port = configuration.getPort();
-		DatagramPacket answerPacket = new DatagramPacket(data, data.length, address, port);
+		DatagramPacket answerPacket = new DatagramPacket(data, len, address, port);
 		sender.send(answerPacket);
 	}
 ```

@Blackjack92 I think it would be best if you decide if the above changes follow the existing concept. If so, let me know and I will commit the changes.

Blackjack92, 2015-12-03:
This problem should be fixed like shown in the afternoon. Pls verify it.

trylimits, 2015-12-06:
Not fixed for other commands. Closing this ticket as Ticket #34 already handles this issue.

closed 2015-12-06 by trylimits