wake-0/fhvRobot #22: [Server] Decline session request if max number of clients reached

trylimits, 2015-12-03:
At the moment the server just stops responding to session requests if there are no more session ids available. The server should send a response message with a declination flag and session id 0.

trylimits, 2015-12-03:
I suggest to handle this case as follows:

To decline a session request, answer `0 0`. Robot code already handles this case correctly.

Blackjack92, 2015-12-03:
This should be implemented pls verify it.

trylimits, 2015-12-07:
I was not able to verify that. I tested it firing >200 session requests to the server and I just did not get any response at all after some time.

trylimits, 2015-12-13:
Works now, except the fact, that session id `0` still shows up in the GUI if all session ids are reserved.

trylimits, 2016-01-03:
I guess this can be considered as "fixed". The fact that clients with invalid session id still show up is ok. It's more like a feature than a bug.

closed 2016-01-03 by trylimits