wake-0/fhvRobot #23: [Server] Duplicate assignment of session ids

trylimits, 2015-12-03:
It seems that the server assigns session ids more than once. See the following screenshot:

[image]

To reproduce this issue just "fire" a bunch of session requests to the server (>5 per second).

I guess this is a multithreading issue of the receive thread.

Blackjack92, 2015-12-03:
Please verify again. The "algorithm" for calculating a new session number has to be improved but for this another ticket should be opened when needed.

trylimits, 2015-12-07:
Just did some testing and it seems that I am able to crash the server when firing a "bunch" of session requests. Looking at the code I guess it's because the server hangs in an endless loop if it fails to find a free session id.

Easy fix: Just implement a safe guard which declines the session request if it fails to get a free session id after 200 tries (should still be fast!). Then we should be fine.

Blackjack92, 2015-12-07:
Pls verify the fix.

trylimits, 2015-12-15:
Seems to work now.

closed 2015-12-15 by trylimits