wake-0/fhvRobot #24: [Server] Mapping from the app to the robo

Blackjack92, 2015-12-03:
Add a robo id to the configuration of a app client. This robo id is 0 at the beginning so the first robo is used. Later it is possible to change this id for each robo which is added.

trylimits, 2015-12-06:
Not 100% clear what you mean by that. Just stating out my ideas:

1. The server manages all registered Robots and Clients
2. The server determines which of the client is the operator of a specific Robot
2.1. An administrator must be able to block a client's operation of a Robot, i.e. a Robot can be "Free to operate" or "Blocked to operate"
2.2. Specific server logic should automatically assign clients to "Free to operate" Robots
3. Only one client can operate a specific Robot at a time
4. The Robot must not interact with the client directly (security issue), i.e. all communication has to be bridged/filtered by the server (already done)

Blackjack92, 2016-01-04:
The filtering should be done via the `OperatorManager`. If it is not correct working open another ticket.

closed 2016-01-04 by Blackjack92