wake-0/fhvRobot #32: [App] Periodical send of commands

Blackjack92, 2015-12-05:
Only send periodical commands to the server or create a bulb of commands, otherwise too much commands are send to the server.

Before this is implemented the latency should be checked when only the real packet is send (at the moment the send packet size is 512 byte).

trylimits, 2015-12-06:
This concept would make sense because of more reasons:

1. Minimizing the network traffic
2. Better response in the App
3. We do not have to care about lost packages as we automatically resend anyway

Blackjack92, 2015-12-07:
Yeah it makes sense but should this not be done on the app? @trylimits 
At the moment to many commands are send because on each progress changed a drive command is send. I will fix this so that only a command is send when it is not the same as before maybe this fix will be enough at the moment.

trylimits, 2015-12-07:
I'm not sure if this implements the desired behavior. We really should implement a timer task which sends commands periodically with a sampling rate of e.g. 5 commands per second. The implemented "simple version" has a weakness in the following case:

1. User pushes up slider to `+100`
2. User pushes down slider to `0`

where 1. and 2. happens in a short time interval. Then packet from 2. may arrive before packet from 1. which leads to an ending state of the robot of `+100`. Even though this case may be not be reproducible, this case will occur according to _Murphy's law_.

 Having a timer task (which runs as thread) will get rid of this issue securely.

Blackjack92, 2015-12-07:
If you send the 5 commands per second the send commands can also be unsorted on the server, the reason is UDP (https://de.wikipedia.org/wiki/User_Datagram_Protocol). As mentioned above it is only a simple fix at the moment which does the desired behavior.

trylimits, 2015-12-07:
I'm aware that UDP does not assure correct order of received packets that's why I. Implementing a periodical send of commands however does absolutely solve the above issue as the send order would be (in worst case):

Packet 2 will arrive before packet 1: Automatic resend after max. 200ms will resend correct packet and bring the Robot to the correct state!

mpe5651, 2016-01-07:
Closed until issue occurs.

closed 2016-01-07 by mpe5651