wake-0/fhvRobot #36: [Server] Heartbeat callback on remove

Blackjack92, 2015-12-06:
When a client is removed because the hearbeat has not come, create a disconnected message.

Blackjack92, 2015-12-08:
We have to decide if this is really necessary.

trylimits, 2015-12-08:
In my opinion this is absolutely necessary. Connection losses are definitely occurring, either because of network problems, low power, forced app-closing or because of other reasons. If we do not recognize those scenarios we face undefined states of App/Server/Robot which is unsatisfying.

Blackjack92, 2015-12-09:
This is not as important as the information when a application message is send and the send client is already disconnected. Because this callback message can be lost because the client is already dc.
I implemented it for the callback the other mentioned type should also be implemented.

closed 2015-12-09 by Blackjack92