wake-0/fhvRobot #40: [Server] Add displaying of stearing controllers

Blackjack92, 2015-12-07:
Add a way for displaying of the stearing controllers.

trylimits, 2015-12-07:
Do you have a proposal?

In my opinion we generally have to options:
Option 1: Complete "joystick" control for JavaFX for mouse control
Option 2: Implement keyboard control (4 keys for controlling left motor up/down and right motor up/down) and show current stearing state with sliders (as in app)

Option 2 will need less time to implement than Option 1.

Blackjack92, 2015-12-07:
At the moment i would suggest to add two scrollbars like in the app and make them readonly, for showing the values from the app. Also a checkbox should be added which enables/disables the control from the app. 

For controlling from the server, the given option via buttons and value textfield is enough at the moment.

trylimits, 2015-12-08:
@Blackjack92 This seems reasonable to me.

Don't you think it would be best if the Robot sends it's actual status to the server so that you can display those values instead of the sent ones? Would be more accurate.

trylimits, 2016-01-04:
In Ticket #52 we already have implemented sliders to control the robot. Above commit additionally gives the possibility to control these sliders with the keys `W` and `S` for left motor up/down, `I` and `K` for right motor up/down and `D` and `J` for resetting the left and right motor.

The implementation uses the JavaFX timeline to animate the sliders after key presses/releases which is not perfect but that way we save one additional keyboard control thread.

closed 2016-01-04 by trylimits