wake-0/fhvRobot #41: [App] Extract configuration (e.g. server address) to static class

trylimits, 2015-12-07:
I reviewed some of the App's code and saw that the IP address of the server is hard-coded into the `NetworkClient` class.

All configuration settings, especially including the Server's IP address should be extracted to a static class `GlobalSettings`, meaningful methods respectively.

Usage should be:

```
String address = GlobalSettings.getServerAddress();
String port = GlobalSettings.getServerPort();
[...]
long samplingRate = GlobalSettinsg.getCommandSamplingRate();
```

This makes our code more maintainable and testable.

Blackjack92, 2015-12-07:
Do you know that our project is still in development phase? That can be the reason that not everyting is fixed at the moment.

closed 2015-12-07 by Blackjack92

trylimits, 2015-12-07:
*I did not intend to offense you and I would like to apologize if I did it unintentionally!*

A ticket system's purpose is to track open issues. Open issues are independent of the project's state and it makes absolutely sense to me to persist open issues at any time someone faces an issue. Also it's not mandatory to fix all these *minor* issues, however, it is important that we keep all those issues in mind!

As you are project lead I leave it up to you to re-open the issue.

Blackjack92, 2015-12-07:
This ticket is closed because I already fixed it.

trylimits, 2015-12-07:
Sorry, haven't seen that :) Thanks, and nice work.