wake-0/fhvRobot #49: [Server] Mediastreaming support

Blackjack92, 2015-12-10:
Add a `DatagramSocket` which is listening to port 999. 
The port should answer with packet size + timestamp.

Blackjack92, 2015-12-10:
The default socket is implemented and listening to incoming data.

trylimits, 2015-12-15:
@Blackjack92 Could you please advice me how I can "push up" the received udp packets in the `MediaStreaming` class to my controller? I'm sure the current design already has something which does that, but I'm still not that familiar with it. I appreciate some hints.

Blackjack92, 2015-12-15:
At the moment there are two different ways. The first one uses a `ClientController` and the second one is delegating over the `NetworkServer` for the `MediaStreaming` no specific implementation is done yet. 
If you want use the same infrastructure as for the `RoboCommunication` and `AppCommunication` you need to extend the `Communication` class.

Blackjack92, 2015-12-15:
But i think it is not necessary to use this `Communication` because then you need to use our defined communication protocol also for the media data.

trylimits, 2015-12-16:
If I got you correctly: The straight forward way would be to inject a callback into the `MediaStreaming` which is called if a packet is received? Seems ok to me.

trylimits, 2016-01-04:
As all related tickets are closed this ticket is also done.

closed 2016-01-04 by trylimits