wake-0/fhvRobot #52: [Server] GUI refurbishment

trylimits, 2015-12-15:
We should refurbish the GUI of the server a bit to provide some more space for the camera view and provide better usability for controlling the robot.

trylimits, 2015-12-15:
Done.

Changes as follows:

1. Got rid of the tab pages
2. Increased size of camera view
3. Showing heartbeat (traffic) in the table view of clients/robots
4. Replaced text boxes for control with sliders
5. Placed sys output at bottom

[image]

Blackjack92, 2015-12-15:
I did the usage of tab pages on purpose. I explained it to you but okay, it seems we have missunderstand ...

Blackjack92, 2015-12-15:
And also the send of specific robo commands is missing in this view. 
What is with the received data? Is it shown in the output below, and can someone see if it is send from the app or the robo?

trylimits, 2015-12-16:
`tfSend` and `btnSend` are still there for Robots and for Apps just below the table views. The button gets enabled if a table view entry is selected.

But you are right: I missed `tfReceive` which is not there anymore. Main reason was probably because I just did not get that it was even there and I do not remember that I ever saw any data in this text field. Nevertheless, we have to further improve/discuss the GUI layout.

The initial purpose of the GUI refurbishment was because of more reasons:

1. There was no space for any camera view
2. Existing GUI elements were too space consuming in relation to their data, e.g. full height for Robot table which contains max. 2 items
3. It was frustrating to debug with Apps and Robots which always required to switch between tabs to see if everything is going right
4. Text field and buttons for (debug-)controlling motors were annoying

closed 2016-01-07 by mpe5651