wake-0/fhvRobot #62: [Server] Reconnect when disconnected

Blackjack92, 2015-12-26:
Reconnect when app is disconnected.

trylimits, 2016-01-06:
At the moment connection loss is reported to the user and the app handles this event correctly by switching to the login screen.

In my opinion the app should not (must not) reconnect automatically/autonomously without letting the user interact. From a usability point of view it would make more sense to let the user decide if (and when) she/he wants to reconnect. This possibility is already provided by the app.

@Blackjack92 What's your opinion on that topic?
@mpe5651 You're also welcome to share your thoughts of course :)

mpe5651, 2016-01-06:
I would also not let the app reconnect automatically. Maybe a reconnect button or something similar in combination with a timeout would address this issue.

Blackjack92, 2016-01-06:
I think that there are two points.
1. It is good to ask the user for reconnect
2. It is bad, when we are in the FHV network and there are always disconnects

mpe5651, 2016-01-07:
after discussion:
- if disconnected, a gray overlay appears
- 10 seconds countdown to automatic reconnect
- tapping to gray overlay starts reconnect immediately
- abort button allows return to login screen

trylimits, 2016-01-09:
Above commit implements this behavior

closed 2016-01-09 by trylimits