wake-0/fhvRobot #64: [App] Usability & Styling

Blackjack92, 2015-12-26:
Styling of the app so that it is more user friendly and easy to use.

trylimits, 2015-12-27:
For usability reasons it is important that the App is running in full "fullscreen", i.e., the notification bar is hidden.

Alternatively, we have to give the sliders more margin at the top.

trylimits, 2016-01-04:
Above commit also (among other things) improves the sliders (more top margin) which allows better control of the robot.

trylimits, 2016-01-04:
Above commit implements snap back of seek bar if released.

trylimits, 2016-01-05:
Above commit implements a "Signal Strength" image in the top right of the app which indicates signal strength (packet loss).

trylimits, 2016-01-05:
Because of the implemented #31 the general style of the App has to be changed to (more modern) Holo theme which made the SeekBars look a little bit weird. We should discuss the exact styling of the SeekBar for controlling the motors.

trylimits, 2016-01-06:
Above commit improves the styling of the app (see commit message).

mpe5651, 2016-01-07:
after discussion:
- add fhv picture while connecting to stream https://upload.wikimedia.org/wikipedia/commons/0/07/Fachhochschule_Vorarlberg_1.jpg

trylimits, 2016-01-10:
I think this can be closed. Style changes should be handled in more specific tickets as basic styling is already done.

closed 2016-01-10 by trylimits