wake-0/fhvRobot #66: [Server] Add server heartbeat

Blackjack92, 2016-01-03:
Add a heartbeat which is send from the server. So every client knows if the server is still online.

Blackjack92, 2016-01-03:
The heartbeat from the server is added. It should be used on the robo and the app @trylimits @mpe5651. This can be done in a different ticket.

trylimits, 2016-01-03:
Could you please summarize the behavior of the heartbeat and what the server expects from the clients? Can I (as robot) just disconnect if I do not receive any heartbeat messages within 5 seconds?

trylimits, 2016-01-03:
Above commit implements disconnect/reconnect if the heartbeat from the server is missing for more than 5s.

We should implement the same/similar behavior for the App client.

Blackjack92, 2016-01-04:
The heartbeat is send from the server to each client (robo and app). The heartbeat contains 
```
as application flags: 0x00
as application command: 0x00
as payload: new byte[] { 0 }
```

It does not expect any response from the clients. You can disconnect when you do not receive any heartbeats for a specific duration (e.g. 5 seconds).

trylimits, 2016-01-05:
The app now handles missing server heartbeats and closes the app safely (with message) when hb is missing for >5s. I also introduced a signal strength "image" depending on the information.

closed 2016-01-05 by trylimits