wake-0/fhvRobot #68: [Server] Slow communication on some machines

trylimits, 2016-01-03:
I observe very slow communication between Server and client instances on my local machine. The problem only occurs if the Robot is connected, i.e. an additional network interface is present.

trylimits, 2016-01-03:
It took some time to track down the problem. I tried several things to solve this issue, including:

1. Replacing lookup of `InetAddress` by getting this information directly from the received packet, i.e. object of type `SocketAddress`
2. Debugging received and sent messages

Finally, I figured out that the method for generating a new session id is causing the issue. To be more precise, the call
```
			newSession = ThreadLocalRandom.current().nextInt(minSession, maxSession);
```
took between 2 and 5 seconds on my machine. Strangely, this only affects the first call. I solved this issue by putting the first call of `ThreadLocalRandom.current()` into a static initializer of the using class.

trylimits, 2016-01-03:
Above commits implement these changes.

trylimits, 2016-01-03:
Above comments and corresponding commits did not entirely solve the problem. The actual cause of the problem was the lookup of the sender's ip address of a `DatagramPacket`.

Some more debugging showed, that the call of `myDatagramPacket.getAddress().getHostName()` caused the performance issue. This call performs a reverse DNS lookup of the sender which can obviously be very slow. See this question on SO: http://stackoverflow.com/questions/11795167/datagrampacket-getaddress-gethostname-is-blocking-my-thread

Replacing these calls according to the answer of the above question solved the issue completely. Above commit implements this.

Blackjack92, 2016-01-04:
Pls check the `DatagramFactory` again there was a bug with `SocketException`.

Blackjack92, 2016-01-04:
@trylimits can this issue be closed again?

trylimits, 2016-01-04:
Yes :)

trylimits, 2016-01-04:
Since this `ThreadLocalRandom` is not supported by android but still causes performance issues, we should replace this call.

trylimits, 2016-01-04:
Done. No performance issues remaining.

closed 2016-01-04 by trylimits