wake-0/fhvRobot #78: [App] compass control

mpe5651, 2016-01-07:


mpe5651, 2016-01-08:
I've done some research about free javafx compass controls. unfortunately, it seems that there is not really much around. I think we'll have to adjust a gauge to a compass. there are some code examples for gauges like http://www.codetown.us/profiles/blogs/javafx-and-steelseries-gauges-using-fxml and libraries like http://www.hameister.org/JavaFX_JFXtrasLib.html which could be useful for our app.

trylimits, 2016-01-11:
After thinking about this issue I think we should not add a conventional compass control like this one:

mpe5651, 2016-01-18:
the line compass should be finished. please verify if it's like you need and then close this ticket.

closed 2016-01-21 by mpe5651