wake-0/fhvRobot #79: [Server] compas control

mpe5651, 2016-01-07:


Blackjack92, 2016-01-09:
I've done some research about free javafx compass controls. unfortunately, it seems that there is not really much around. I think we'll have to adjust a gauge to a compass. there are some code examples for gauges like http://www.codetown.us/profiles/blogs/javafx-and-steelseries-gauges-using-fxml and libraries like http://www.hameister.org/JavaFX_JFXtrasLib.html which could be useful for our app.

Also interesting: https://bitbucket.org/hansolo/enzo/overview

Blackjack92, 2016-01-09:
https://bitbucket.org/hansolo/airseries

trylimits, 2016-01-11:
After thinking about this issue I think we should not add a conventional compass control like this one:
https://lh4.ggpht.com/hpp9n8LN6Qc9n8labOZ-CLpfqf8LAe3D94GpYN4iQYTFH0zM82h6onWSF5nQY9TRCgj8=w300
Instead we should add the bottom of the view a widget like this (as known from some games):
http://www.pcgameshardware.de/screenshots/original/2015/11/FO4_Kompass_Fix-pcgh.jpg

In my opinion this will be more "first person viewish".

trylimits, 2016-01-18:
I have implemented a simple 3D view which shows the actual values for yaw, roll and pitch orientation for testing the sensor values. I will try to integrate this view into the server application with low effort. I think this is a good replacement for a compass view.

trylimits, 2016-01-18:
Above commit adds the 3d view as overlay to the camera view. The 3d model is a (very) simplified version of our robot. The 3d world also contains the x,y and z axis. There is no binding of orientation values to the 3d model yet. This should be implemented in another ticket as it requires extending the model of the client.

closed 2016-01-18 by trylimits