wake-0/fhvRobot #88: [Game] Game concept

Blackjack92, 2016-01-09:
Think about a game concept e.g. game time, processing time between the players,  ....

mpe5651, 2016-01-11:
since we have a consensus about offering a parkour, here are my thoughts:

to the parkour itself: 
I would suggest to build it out of wood and polystyrene (=> styropor) as this allows us to easily build a roadway with side boards. the side boards prevent the robot from leaving the roadway. with the polystyrene components of the parkour such as up and down ramps, obstacles which have to be bypassed and tunnels are easily made. an additional idea would be to arrange one part of the roadway to be slippery(e.g. with sand, water or even with oil) in order to make it more challenging.

to the game itself:
the parkour itself will be organised cirularly in order to have the starting and the end point at the same location.
two different modes should be at disposal:

1.) free ride: in this mode the user gets for a defined time, lets say 90 oder 120 seconds, to try out the parkour and get a feeling for the steering over the app. during this period, triggering of the proposed light barrier is simply ignored. after this time expires, the user of the app can choose between driving a race or leaving. during the free ride a premature handover/leaving should also be possible(let's say a button or an back arrow above of the left steering pad). in any case, we need a wait time of about 20-30 seconds in order to manually bring the robot back to the starting point. this is necessary since we have no automatic return-to-start. I thought a lot of this issue with the wait time, but I'm afraid I have no better idea at the moment to offer how to make it more elegant.

2.) race: when this mode is choosen, the time measurement is started with a light barrier. we can rape the lidar and use it as light barrier, or use right from the start a real one. anyway, when the time measurement is started, the user has a maximum of 4 minutes to complete the parkour. the race ends when the robot triggers the light barrier for the second time. then the name of the player and his time together with a ranking are displayed by the game server. in case of time expiration or premature handover the wait time of 20-30 seconds is needed to bring the robot back to the starting point. 

on premature handover, I would suggest to let the next user connect only AFTER the wait time has expired. by doing so the user does not have to wait for the robo to be brought back to starting position. in case of first driving a free ride and then wanting to drive a race, a gray overlay window with a countdown timer and an appropriate message should be displayed.

grading system: the only grading system by now is the time. the fastest one is the first in the ranking. and the ones who leave prematurely or need longer than the maximum time should automatically queued as the last ones in the ranking.

that should be all for now, if I haven't forgot anything. the appearance of the parkour is something I already have in mind and want to show to you on thursday, 14.01.16.

trylimits, 2016-01-12:
Good work @mpe5651

closed 2016-01-21 by mpe5651