wake-0/fhvRobot #90: [Game Server] Implement Highscore list

trylimits, 2016-01-10:
We have to implement a Highscore system. The class which is responsible for this functionality should at least provide the following methods:

`void SaveToHighscore(String name, TimeSpan time);`
`List<HighscoreItem> LoadHighscore();`
`String GetXMLData();`
`void SaveToXMLFile(String filename);`
`void LoadFromXMLFile(String filename);`

@Blackjack92 could you add/change specification acc your opinion and implement this class/service.

Blackjack92, 2016-01-10:
Yeah I will think about a concept and implement it.

Blackjack92, 2016-01-10:
@trylimits there will be a problem when the data is send to the server because the max size of the application payload is defined by 1 byte (See https://github.com/Blackjack92/fhvRobot/wiki/Protocol-definitions). 

So only 255 bytes can be send even less when the conversion from byte to int on the server is incorrect we have to check this.

trylimits, 2016-01-11:
That's correct. One possibility would be to define a new command which explicitly overrides the payload field with a length of 2 bytes which would be enough or use the version-bits in the application layer's flag byte.

Blackjack92, 2016-01-11:
http://stackoverflow.com/questions/1936857/convert-integer-into-byte-array-java

Blackjack92, 2016-01-11:
See ticket #91.

Blackjack92, 2016-01-12:
The highscore list is implemented. The send of this list should be handled in a different ticket.

closed 2016-01-12 by Blackjack92