wake-0/fhvRobot #96: [Game Server] Define interface for time measurement system

trylimits, 2016-01-13:
One should define an interface `ITimeMeasurementSystem` for the time measurement system in the Game Server project.

The contract between the main controller and the time measurement should be as follows:

1. Controller can set the time measurement system active
2. Controller can set the time measurement system inactive
3. The time measurement system can raise an event (or callback) on the controller that a player has triggered time measurement

The controller should behave correctly if the event (3.) is raised.
The time measurement system MUST NOT raise any event if the system is set to inactive mode.

Blackjack92, 2016-01-13:
@trylimits pls have a look at the interface.

Blackjack92, 2016-01-13:
Above commit implements the ITriggerSystem like discussed.

closed 2016-01-13 by Blackjack92