wake-0/fhvRobot #107: [App] reconnect count down on app closure

mpe5651, 2016-01-18:
there is a reproducable bug (at least on my htc one m7): when "Abbrechen" is pressed, the app closes, but the reconnect time out is left to count down.

[image]

trylimits, 2016-01-18:
Is there any exception thrown?
What happens if you press _back_ instead of "Abbrechen"?

mpe5651, 2016-01-18:
no, there is no exception at all. there is only a message "RoboApp wurde beendet" before the timer is displayed. pressing back or "abbrechen" stops the timer and you are in your phone's home menu. the most weird thing is that this app only occurs on my htc phone.

trylimits, 2016-01-21:
Maybe this was because of #126 . Could you please try again with the newest version?

trylimits, 2016-01-24:
@mpe5651 Could you please re-check if the bug still exists in the newest version?

trylimits, 2016-01-28:
Acc. to @mpe5651 this bug is fixed.

closed 2016-01-28 by trylimits