wake-0/fhvRobot #127: [Game Server] implement camera time measurement

mpe5651, 2016-01-21:
implement image processing for visually detecting when to start/stop time measurement

trylimits, 2016-01-24:
I have implemented the camera time trigger system and already integrated it in the settings view.

The camera trigger system has the following features:

0. Definable region of interest (ROI)
1. Trigger threshold can be defined
2. Self-adapting on environmental changes (light ...)
3. Block time between to triggers (10s by default) can be defined to prevent multiple triggers if the robot passes the start line slowly
4. The trigger system has a GUI (for debugging and specifying the ROI)

@Blackjack92 I have integrated the new system as follows. In the `SettingsViewModel` I applied these changes:

```
~~TriggerSystem = new TriggerSystemMock();~~
TriggerSystem = CameraTriggerService.Instance;
```

closed 2016-01-24 by trylimits

Blackjack92, 2016-01-25:
@trylimits pls see the code refactoring. 
If you set the content of the toggle button, it will not change the text when the toggle state is changed. Normally we add to a command the postfix `Command`.

To the `TimeMeasurementSystem` a hint use a OnTriggerRaised Method.
E.g. from msdn (https://msdn.microsoft.com/en-us/library/aa645739(v=vs.71).aspx)
```
// A class that works just like ArrayList, but sends event
   // notifications whenever the list changes:
   public class ListWithChangedEvent: ArrayList 
   {
      // An event that clients can use to be notified whenever the
      // elements of the list change:
      public event EventHandler Changed;

      // Invoke the Changed event; called whenever list changes:
      protected virtual void OnChanged(EventArgs e) 
      {
         if (Changed != null)
            Changed(this,e);
      }

      // Override some of the methods that can change the list;
      // invoke event after each:
      public override int Add(object value) 
      {
         int i = base.Add(value);
         OnChanged(EventArgs.Empty);
         return i;
      }

      public override void Clear() 
      {
         base.Clear();
         OnChanged(EventArgs.Empty);
      }

      public override object this[int index] 
      {
         set 
         {
            base[index] = value;
            OnChanged(EventArgs.Empty);
         }
      }
   }
```