wake-0/fhvRobot #131: [Game Server] auto-scroll highscore

mpe5651, 2016-01-21:


mpe5651, 2016-01-21:
I've done some research on this issure. there are some articles which seem to be very promissing, especially the following three:
- http://stackoverflow.com/questions/8480252/is-there-a-simple-way-to-have-a-listview-automatically-scroll-to-the-most-recent
- http://stackoverflow.com/questions/3317194/mvvm-how-to-make-a-list-view-auto-scroll-to-a-new-item-in-a-list-view
- http://stackoverflow.com/questions/2006729/how-can-i-have-a-listbox-auto-scroll-when-a-new-item-is-added

I'll try to get one of the suggested solutions to work.

Blackjack92, 2016-01-24:
I have reviewed your code @mpe5651. I would suggest to create a own control. So we put the scroll behavior to the control which should scroll not to the entire view/window
```
public class ListBoxScroll : ListBox
{
    public ListBoxScroll() : base()
    {
        SelectionChanged += new SelectionChangedEventHandler(ListBoxScroll_SelectionChanged);
    }

    void ListBoxScroll_SelectionChanged(object sender, SelectionChangedEventArgs e)
    {
        ScrollIntoView(SelectedItem);
    }
}
```

Blackjack92, 2016-01-25:
@mpe5651 
At the moment it seems that the listview only scrolls to the end of the list not to the current added item, is this right? So i think we have to use the parameters (`NotifyCollectionChangedEventArgs`) to get the added item and scroll to this item.

trylimits, 2016-01-25:
I have implemented a continuous automatic scrolling behavior of the `ListView` containing ranks 3 to x. Looks good.

closed 2016-01-25 by trylimits