wake-0/fhvRobot #140: [Game Server] Connect TimeTriggerSystem with duration of player

Blackjack92, 2016-01-25:
The ``TimeTriggerSystem` should be connected to the duration of the player. Also the `TimerService` needs is part of this ticket.

trylimits, 2016-01-25:
Above commit connects the `TriggerSystem` to the `TimerService`. If a trigger is raised the clock starts/stops, however, no entries are added to the list yet.

trylimits, 2016-01-25:
Above commit implements the following:

1. Trigger system is connected to timer service
2. Finish trigger sets time to `ScoreManager.CurrentScore`
3. Currently best time of the player is shown in the GUI
4. Settings window contains a button to explicitly save the current best score of the player
5. If 4. is executed the currently best score is discarded

I foresaw some problems if implementing automatic saving of the best score at each player switch. We have to do some testing with that and can then implement this if needed.

closed 2016-01-25 by trylimits