wake-0/fhvRobot #182: [Documentation] [Wiki]

trylimits, 2016-02-06:
We have to document our work until 15-02-2016. As discussed we should use the public wiki of our project's github page to document the work.

Additionally, for handing in the documentation we have to generate a pdf version of the wiki. The wiki itself is written in markdown, therefore we have to export those markdown files into a pdf. In order to do that (using pandoc) we have to follow these guidelines:

1. `git clone` the project's wiki
2. Change the markdown files locally and push frequently (we can use a markdown editor)
3. File names have to be camel-cased (no blanks/spaces)
4. After a chapter we have to use `\pagebreak` (not interpreted by github, but mandatory for exporting correctly)
5. to be continued (anyone is invited to share his knowledge)

trylimits, 2016-02-06:
I already updated the wiki and added a proposal for a rough concept of the documentation (inc. assignees for the chapters). Changes are welcome.

Blackjack92, 2016-02-09:
It is necessary to discuss the large and lower case of the words app, robo, server, game server ...